stephenlacy/msi-keyboard #11: How would I go about defining new colors?

asonix, 2015-02-04:
I think the title says enough. I tracked the colors to keyboard.sendFeatureReport(activate), but I'm not sure where the code that describes this comes from.

stevelacy, 2015-02-04:
The array of colors are defined here: https://github.com/wearefractal/msi-keyboard/blob/master/lib/constants.json

There is not any spec from SteelSeries for the actual color codes, just reverse-engineered numbers.

asonix, 2015-02-04:
I have done some testing and found more colors

color             #    brightness
red2             9    all
green2         10  all
blue2           11  all
white2          12  all
aqua            23  all
cyan             24  all
blue3            25  all
magenta       26  low
purple2         26  med
cyan2           26  high
red3             27  low, med
white3          27  high
yellow2        28  low
orange2      28  med
pink2           28  high
blue-white    29  low, med
yellow-white  29 high
green2          30 low
yellow3          30 med
white4           30 high
purple3         31 low
dark-purple   31 med
green3          31 high
red4              32 low
off                 32 med
blue4            32 high
white5           33 low
pink3             33 med
red5              33 high
off                 34 low
red6              34 med
sky2              34 high

I'll expand my script past 40 and see if there's anything else

stevelacy, 2015-02-04:
Yes, I searched all the color numbers but kept just the primary colors from the window app in the project.

stevelacy, 2015-03-14:
Closing this, if you want to send a PR for the colors list - feel free.

closed 2015-03-14 by stevelacy