stephenlacy/msi-keyboard #12: cannot open device with vendor id 0x1770 and product id 0xff00 although using sudo

renry, 2015-02-14:
Hi,

I use the command:
sudo node server.js (or sudo nodejs server.js)

but It returned me:
cannot open device with vendor id 0x1770 and product id 0xff00

Any suggestions?

Thanks in advance

renry, 2015-02-14:
My laptot is an MSI GE60

stevelacy, 2015-02-14:
Did you install the dependencies listed on the readme?

renry, 2015-02-15:
I have installed libusb-dev and libusb-1.0-0-dev
and
npm install

renry, 2015-02-15:
today I installed linux mint replacing ubuntu, but there is the same problem:
cannot open device with vendor id 0x1770 and product id 0xff00

stevelacy, 2015-02-16:
That is a permission error related to either a missing package, or not using sudo.

renry, 2015-02-16:
this is the correct README you said?
https://github.com/wearefractal/msi-keyboard/blob/master/README.md

stevelacy, 2015-02-16:
Yes, on linux mint I was only required to install those two libusb libraries.

au-phiware, 2015-04-24:
I am also getting this error while using sudo. I have Gentoo (kernel v3.19.3) on a GS70 2QE, I emerge'd dev-libs/libusb-1.0.19 (which installs both libusb and libusb-dev). I simply cloned this repository and ran:

    npm install
    sudo npm test

...two of the tests throw "cannot open device with vendor id 0x1770 and product id 0xff00" and the other passes.

I can see the device using `lsusb` but some information is missing:

````
$ sudo lsusb -d 1770:ff00 -v

Bus 004 Device 003: ID 1770:ff00  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1770 
  idProduct          0xff00 
  bcdDevice            1.10
  iManufacturer           1 MSI EPF USB
  iProduct                1 MSI EPF USB
  iSerial                 1 MSI EPF USB
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0
      bInterfaceProtocol      0
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.10
          bCountryCode           33 US
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      57
          Report Descriptor: (length is 57)
            Item(Global): Usage Page, data= [ 0xa0 0xff ] 65440
                            (null)
            Item(Local ): Usage, data= [ 0x01 ] 1
                            (null)
            Item(Main  ): Collection, data= [ 0x01 ] 1
                            Application
            Item(Local ): Usage, data= [ 0x02 ] 2
                            (null)
            Item(Main  ): Collection, data= [ 0x00 ] 0
                            Physical
            Item(Global): Usage Page, data= [ 0xa1 0xff ] 65441
                            (null)
            Item(Global): Report ID, data= [ 0x01 ] 1
            Item(Local ): Usage, data= [ 0x03 ] 3
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Physical Minimum, data= [ 0x00 ] 0
            Item(Global): Physical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x07 ] 7
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Global): Report ID, data= [ 0x02 ] 2
            Item(Local ): Usage, data= [ 0x04 ] 4
                            (null)
            Item(Global): Logical Minimum, data= [ 0x00 ] 0
            Item(Global): Logical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Physical Minimum, data= [ 0x00 ] 0
            Item(Global): Physical Maximum, data= [ 0xff 0x00 ] 255
            Item(Global): Report Size, data= [ 0x08 ] 8
            Item(Global): Report Count, data= [ 0x00 0x01 ] 256
            Item(Main  ): Feature, data= [ 0x02 ] 2
                            Data Variable Absolute No_Wrap Linear
                            Preferred_State No_Null_Position Non_Volatile Bitfield
            Item(Main  ): End Collection, data=none
            Item(Main  ): End Collection, data=none
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              10
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)
````

Can you help?

stevelacy, 2015-04-24:
Can you create a `node-hid` script to identify the vendorId is correctly connecting to the device?

```js
var hid = require('node-hid');
console.log(hid.devices());

var board = new hid.HID(6000, 65280);
console.log(board);
```
`sudo node hid.js`

au-phiware, 2015-04-26:
Yes, it appears to connect to the device:

````
$ sudo node hid.js
[ { vendorId: 6000,
    productId: 65280,
    path: '0004:0004:00',
    serialNumber: 'MSI EPF USB',
    manufacturer: 'MSI EPF USB',
    product: 'MSI EPF USB',
    release: 272,
    interface: 0,
    usagePage: 0,
    usage: 0 } ]
{ domain: null,
  _events: { newListener: [Function] },
  _maxListeners: 10,
  _raw: {},
  read: [Function],
  write: [Function],
  getFeatureReport: [Function],
  sendFeatureReport: [Function],
  setNonBlocking: [Function],
  _paused: true }
````

au-phiware, 2015-04-26:
I should note that I don't even have the default red backlight on my keyboard...

au-phiware, 2015-04-26:
I do not know why but this works for me:

	var hid = require('node-hid');
	var constants = require('./lib/constants');
	var keyboard = new hid.HID(6000, 65280);

	var color = constants.colors.green;
	var commit = new Buffer(8);
	commit[0] = 1;
	commit[1] = 2;
	commit[2] = 65; // commit
	commit[3] = constants.modes.normal; // set hardware mode
	commit[4] = 0;
	commit[5] = 0;
	commit[6] = 0;
	commit[7] = 236; // EOR

	var activate = new Buffer(8);
	activate[0] = 1;
	activate[1] = 2;
	activate[2] = 66; // set
	activate[3] = constants.regions.left;
	activate[4] = color;
	activate[5] = constants.levels.med;
	activate[6] = 0;
	activate[7] = 236;  // EOR (end of request)
	keyboard.sendFeatureReport(activate);

	activate[0] = 1;
	activate[1] = 2;
	activate[2] = 66; // set
	activate[3] = constants.regions.middle;
	activate[4] = color;
	activate[5] = constants.levels.med;
	activate[6] = 0;
	activate[7] = 236;  // EOR (end of request)
	keyboard.sendFeatureReport(activate);

	activate[0] = 1;
	activate[1] = 2;
	activate[2] = 66; // set
	activate[3] = constants.regions.right;
	activate[4] = color;
	activate[5] = constants.levels.med;
	activate[6] = 0;
	activate[7] = 236;  // EOR (end of request)
	keyboard.sendFeatureReport(activate);
	keyboard.sendFeatureReport(commit);

But `sudo npm test` still fails.

au-phiware, 2015-04-26:
I am able to replicate the error with this code:

	var hid = require('node-hid');
	var keyboard = new hid.HID(6000, 65280);
	keyboard = new hid.HID(6000, 65280);

renry, 2015-05-04:
For me "sudo lsusb" return:
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 13d3:3394 IMC Networks 
Bus 003 Device 004: ID 1235:800a Novation EMS 
Bus 003 Device 003: ID 0763:0198 Midiman Axiom 25
Bus 003 Device 002: ID 046d:c050 Logitech, Inc. RX 250 Optical Mouse
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I can't see keyboard.

with "sudo lsusb -d 1770:ff00 -v"
return nothing.

Any suggestions?

stevelacy, 2016-05-02:
#40

closed 2016-05-02 by stevelacy