stephenlacy/msi-keyboard #13: "Error: Module did not self-register" when using node v0.12

alicee, 2015-02-24:
When running `sudo node examples/sample.js` using node version v0.12.0, I get the following error message:

```
module.js:355
  Module._extensions[extension](this, filename);
                               ^
Error: Module did not self-register.
    at Error (native)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Module.require (module.js:365:17)
    at require (module.js:384:17)
    at Object.<anonymous> (/home/alice/config/msi-keyboard/node_modules/node-hid/index.js:5:15)
    at Module._compile (module.js:460:26)
    at Object.Module._extensions..js (module.js:478:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
```

Everything works fine using v0.10.35.

stevelacy, 2015-02-24:
Rebuild the `node-hid` module with node v0.12.0:
`rm -rf node_modules && npm install`

alicee, 2015-02-24:
That fails unfortunately, but as far as I can tell the issue is with node-hid or one of its dependencies: https://github.com/node-hid/node-hid/issues/90

stevelacy, 2015-03-14:
@alicee When they fix it, could you comment here?

alicee, 2015-03-17:
@stevelacy Yes, I'll update you once there's a fix.

demiansan, 2015-04-03:
A little trick to turn on msi keyboard lights is pressing two or more times the msi light button (the one that changes illumination program in windows), only white colour is activated but at least is on.

Works on my laptop: MSI GE60 PC Apache; maybe it will work on other models.

stevelacy, 2015-04-03:
@demiansan the issue is to do with node v12 and `node-hid`, not if the hid device was turned on.

demiansan, 2015-04-03:
@stevelacy Absolutely right, I am sorry for avoiding the primary objective of this issue.

stevelacy, 2015-04-03:
@demiansan it's fine, if you want to send a PR to add that helping hint to the readme, feel free :)

stevelacy, 2016-05-02:
#40

closed 2016-05-02 by stevelacy