stephenlacy/msi-keyboard #14: MSI Stealth

phillw, 2015-03-13:
Hi,

First - thank you, I have been waiting over a year for this :-)

For the MSi Stealth GS70 series, only the blink example seams to work but nothing else. Unable to change colours or anything.

I have no errors to show, as there is no output other than setColor:

describe('msi-keyboard', function() {
^
ReferenceError: describe is not defined
    at Object.<anonymous> (/home/phill/workspace/keyboard/test/setColor.js:6:1)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3

Hope you can help,
Best Regards,

stevelacy, 2015-03-13:
That error is expected, it is not an example file but a code test.

I have the keyboard in the GS70 Stealth, please run this [example](https://github.com/wearefractal/msi-keyboard/blob/master/examples/blinkMulti.js) with sudo.

phillw, 2015-03-13:
Yes, the blinking works perfectly. But none other works.
No wave, etc. the example to set colour.

Ideally would like to have wave when computer is idle otherwise just have
non blinking/moving colours.

phillw, 2015-03-13:
Hi,

Thanks for the fast response.
It is running as root / sudo.

The blinking example work perfectly :-) however wave / breath / just colour set has no output.

stevelacy, 2015-04-08:
The issue with wave and breath will be fixed with #20

closed 2016-02-08 by stevelacy