stephenlacy/msi-keyboard #15: Linux Mint 17.1 64bit instructions

bash64, 2015-03-14:
linux steel engine keyboard

Problem: You own a MSI gaming laptop. You would like to have the Steel Series Keyboard light up with those pretty rainbow colors and breathe and do the wave or just glow.

The following is lengthy but so simple a child could do it. I put in many hours piecing together various bits of javascript code from different places to make this all happen as well as spending many hours into the night writing and testing my documentation for Linux Mint 17.1 64bit. 

I own a MSI GT70 2PE. My usb keyboard shows up as such:
>lsub
Bus 003 Device 003: ID 1770:ff00

If you see something different then I am not at a point to help you.
You'd have to modify the javacode and no guarantee this will work.
Many report this as working on several MSI laptops.

ok, here we go. grab some soda and chips, your in for a ride:

MSI Steel engine keyboard javascript commands for Linux
------------------------------------------------------------------------------

Here are the websites I am used for reference in case it won't work for you.

http://forum.techinferno.com/msi/5269-msi-led-controller-linux-windows-mac-|-msi-klm-alternative-2.html

https://www.npmjs.com/package/node-hid

https://github.com/wearefractal/msi-keyboard/issues/1

steps:

#enter your home folder
>cd ~

#become root
>sudo bash

#do the following. just cut and paste each step in a terminal...
>apt-get install libusb-dev libusb-1.0-0-dev
>apt-get install nodejs npm
>git clone https://github.com/wearefractal/msi-keyboard
>git clone https://github.com/hanshuebner/node-hid
>cd node-hid; npm install -g
>cd ~
>cd msi-keyboard; npm install --g

# create the following breathe.js file and put it anywhere you want to....paste in the really huge javascript code provided below that says it is for breathe and wave...
>sudo vi breathe.js 

# now run the javascript...
>sudo nodejs breathe.js

#your keyboard will breathe....

# make a copy of this script to do the wave effect..
>sudo cp breathe.js wave.js

# In order to make the keyboard do a wave effect modify the wave.js file slightly by
# commenting out (// = comment out) the breathe(1, 2, 3, 3, 1); line and uncomment the wave(1, 2, 3, 3, 1); line...
# both of these lines are at the very top of the script.
>sudo vi wave.js

# now do the wave
>sudo nodejs wave.js

#Take the following 4 lines of javascript code and save it as rainbow.js
# This will just make the keyboard light up with red/green/blue without doing any effects.
>sudo vi rainbow.js

var keyboard = require('/usr/local/lib/node_modules/msi-keyboard');
keyboard.color('left',{color: 'red', intensity: 'med'});
keyboard.color('middle',{color: 'blue', intensity: 'med'});
keyboard.color('right',{color: 'green', intensity: 'med'});

# light her up...
>sudo nodejs rainbow.js

#if you get a usb error such as 'vendor id #xxxx and product is# not found' then you forgot to use sudo. usb access needs root privileges.

#-----------------------------------------------------------------------------------------------------#
#The following is the really huge javascript code mentioned above to make your keyboard BREATHE or WAVE.
# It begins with the 'var' statement. Do not paste in this boxed section:
#-----------------------------------------------------------------------------------------------------#

var keyboard = require('/usr/local/lib/node_modules/msi-keyboard');

// the breathe line
breathe(1, 2, 3, 3, 1); // COLOR1, COLOR2, COLOR3, PERIOD, LEVEL

// the wave line
//wave(1, 2, 3, 3, 1);

// possible values:
// colors: 0-8
// period: the unit might be seconds?
// level: 0-3

function disable() {
    commit(0);
}

function wave(color1, color2, color3, period, level) {
    // Convert colors to RGB
    color1RGB = getRGB(color1);
    color2RGB = getRGB(color2);
    color3RGB = getRGB(color3);

    // Create three new "speed colors"
    var speedColor1 = { r:0, g:0, b:0 };
    var speedColor2 = { r:0, g:0, b:0 };
    var speedColor3 = { r:0, g:0, b:0 };

    speedColor1.r = computeRampSpeed(color1RGB.r, 0, period);
    speedColor1.g = computeRampSpeed(color1RGB.g, 0, period);
    speedColor1.b = computeRampSpeed(color1RGB.b, 0, period);

    speedColor2.r = computeRampSpeed(color2RGB.r, 0, period);
    speedColor2.g = computeRampSpeed(color2RGB.g, 0, period);
    speedColor2.b = computeRampSpeed(color2RGB.b, 0, period);

    speedColor3.r = computeRampSpeed(color3RGB.r, 0, period);
    speedColor3.g = computeRampSpeed(color3RGB.g, 0, period);
    speedColor3.b = computeRampSpeed(color3RGB.b, 0, period);

    // Send data
    sendActiveArea(0x43, 1, color1, level, 0);
    sendActiveArea(0x43, 2, 0, level, 0);
    sendActiveArea(0x43, 3, speedColor1.r, speedColor1.g, speedColor1.b);

    sendActiveArea(0x43, 4, color2, level, 0);
    sendActiveArea(0x43, 5, 0, level, 0);
    sendActiveArea(0x43, 6, speedColor2.r, speedColor2.g, speedColor2.b);

    sendActiveArea(0x43, 7, color3, level, 0);
    sendActiveArea(0x43, 8, 0, level, 0);
    sendActiveArea(0x43, 9, speedColor3.r, speedColor3.g, speedColor3.b);

    commit(5);
}

function breathe(color1, color2, color3, period, level) {
    // Convert colors to RGB
    color1RGB = getRGB(color1);
    color2RGB = getRGB(color2);
    color3RGB = getRGB(color3);

    // Create three new "speed colors"
    var speedColor1 = { r:0, g:0, b:0 };
    var speedColor2 = { r:0, g:0, b:0 };
    var speedColor3 = { r:0, g:0, b:0 };

    speedColor1.r = computeRampSpeed(color1RGB.r, 0, period);
    speedColor1.g = computeRampSpeed(color1RGB.g, 0, period);
    speedColor1.b = computeRampSpeed(color1RGB.b, 0, period);

    speedColor2.r = computeRampSpeed(color2RGB.r, 0, period);
    speedColor2.g = computeRampSpeed(color2RGB.g, 0, period);
    speedColor2.b = computeRampSpeed(color2RGB.b, 0, period);

    speedColor3.r = computeRampSpeed(color3RGB.r, 0, period);
    speedColor3.g = computeRampSpeed(color3RGB.g, 0, period);
    speedColor3.b = computeRampSpeed(color3RGB.b, 0, period);

    // Send data
    sendActiveArea(0x43, 1, color1, level, 0);
    sendActiveArea(0x43, 2, 0, level, 0);
    sendActiveArea(0x43, 3, speedColor1.r, speedColor1.g, speedColor1.b);

    sendActiveArea(0x43, 4, color2, level, 0);
    sendActiveArea(0x43, 5, 0, level, 0);
    sendActiveArea(0x43, 6, speedColor2.r, speedColor2.g, speedColor2.b);

    sendActiveArea(0x43, 7, color3, level, 0);
    sendActiveArea(0x43, 8, 0, level, 0);
    sendActiveArea(0x43, 9, speedColor3.r, speedColor3.g, speedColor3.b);

    commit(3);
}

// The rest are help functions

function sendActiveArea(modeValue, area, color, level, blue) {
    var data = new Buffer(8);

    data[0] = 0x01;
    data[1] = 0x02;
    data[2] = modeValue;
    data[3] = area;
    data[4] = color;
    data[5] = level;
    data[6] = blue;
    data[7] = 0xec;

    keyboard.sendFeatureReport(data);
}

function getRGB(color) {
    var rgb = {r:0, g:0, b:0};

    if (color == 0) {       // BLACK
        rgb.r = 0;
        rgb.g = 0;
        rgb.b = 0;
    } else if (color == 1) {    // RED
        rgb.r = 255;
        rgb.g = 0;
        rgb.b = 0;
    } else if (color == 2) {    // ORANGE
        rgb.r = 187;
        rgb.g = 112;
        rgb.b = 0;
    } else if (color == 3) {    // YELLOW
        rgb.r = 238;
        rgb.g = 238;
        rgb.b = 0;
    } else if (color == 4) {    // GREEN
        rgb.r = 176;
        rgb.g = 255;
        rgb.b = 0;
    } else if (color == 5) {    // SKY
        rgb.r = 0;
        rgb.g = 255;
        rgb.b = 255;
    } else if (color == 6) {    // BLUE
        rgb.r = 0;
        rgb.g = 0;
        rgb.b = 255;
    } else if (color == 7) {    // PURPLE
        rgb.r = 48;
        rgb.g = 0;
        rgb.b = 255;
    } else if (color == 8) {    // WHITE
        rgb.r = 176;
        rgb.g = 255;
        rgb.b = 176;
    }

    return rgb;
}

function computeRampSpeed(leftColor, rightColor, period) {
    if (leftColor - rightColor == 0) {
        return 0;
    }

    return Math.ceil((period * 250) / Math.abs(leftColor - rightColor));
}

function commit(mode) {
    var data = new Buffer(8);

    data[0] = 0x01;
    data[1] = 0x02;
    data[2] = 0x41;
    data[3] = mode;
    data[4] = 0x00;
    data[5] = 0x00;
    data[6] = 0x00;
    data[7] = 0xec;

    keyboard.sendFeatureReport(data);
}

Toadrill, 2015-08-08:
if the npm install fails you may need to install nodejs-legacy
see [this page](http://stackoverflow.com/questions/21168141/can-not-install-packages-using-node-package-manager-in-ubuntu) for explanation

bash64, 2015-08-10:
Here are my instructions for Linux Mint 64 bit 17.2 cinnamon:

Status: Verified and fully tested.

MSI steel engine keyboard

Problem: You own a MSI gaming laptop. You would like to have the Steel Series Keyboard light up with those pretty rainbow colors and breathe and do the wave or just glow.

The following is lengthy but so simple a child could do it. I put in many hours piecing together various bits of javascript code from different places to make this all happen as well as spending many hours into the night writing and testing my documentation for Linux Mint 17.1 64bit. 

I own a MSI GT70 2PE. My usb keyboard shows up as such:
>lsusb
Bus 003 Device 003: ID 1770:ff00

If you see something different then I am not at a point to help you.
You'd have to modify the javacode and no guarantee this will work.
Many report this as working on several MSI laptops.

ok, here we go. grab some soda and chips, your in for a ride:

MSI Steel engine keyboard javascript commands for Linux
------------------------------------------------------------------------------

Here are the websites I am using for reference in case it won't work for you.

http://forum.techinferno.com/msi/5269-msi-led-controller-linux-windows-mac-|-msi-klm-alternative-2.htmlhttps://www.npmjs.com/package/node-hidhttps://github.com/wearefractal/msi-keyboard/issues/1

steps:
If you reinstall your OS, you will lose the modules in the /usr folder and have to do this again even if you keep your home folder. Just start over from scratch.

#open a terminal and enter your home folder
>cd ~

#elevate your privileges...
>sudo bash

Do the following. Just cut and paste each step into your terminal...
(Obviously the '>' sign does not get copied and pasted....

>apt-get install git libusb-dev libusb-1.0-0-dev nodejs npm
>ln -s /usr/bin/nodejs /usr/bin/node
>git clone https://github.com/wearefractal/msi-keyboard
>git clone https://github.com/hanshuebner/node-hid
>cd ~/node-hid
>git clone https://github.com/signal11/hidapi.git
>cd ~/node-hid; npm install -g
>cd ~/msi-keyboard; npm install --g

Create the following breathe.js file and put it anywhere you want to....paste in the really huge javascript code provided below that says it is for breathe and wave...
>sudo nano breathe.js 

Now run the javascript...
>sudo nodejs breathe.js

Your keyboard will breathe....

Make a copy of this script to do the wave effect..
>sudo cp breathe.js wave.js

In order to make the keyboard do a wave effect modify the wave.js file slightly by
commenting out (// = comment out) the breathe(1, 2, 3, 3, 1); line and uncomment the wave(1, 2, 3, 3, 1); line...
both of these lines are at the very top of the script.
>sudo nano wave.js

Now do the wave
>sudo nodejs wave.js

Take the following 4 lines of javascript code and save it as rainbow.js
This will just make the keyboard light up with red/green/blue without doing any effects.
>sudo nano rainbow.js
[... 169 lines cut ...]
    return rgb;
}

function computeRampSpeed(leftColor, rightColor, period) {
    if (leftColor - rightColor == 0) {
        return 0;
    }

    return Math.ceil((period * 250) / Math.abs(leftColor - rightColor));
}

function commit(mode) {
    var data = new Buffer(8);

    data[0] = 0x01;
    data[1] = 0x02;
    data[2] = 0x41;
    data[3] = mode;
    data[4] = 0x00;
    data[5] = 0x00;
    data[6] = 0x00;
    data[7] = 0xec;

    keyboard.sendFeatureReport(data);
}

stevelacy, 2015-08-10:
I will close + lock this issue as it has not turned into a PR and is not relating to any issue

closed 2015-08-10 by stevelacy