stephenlacy/msi-keyboard #18: Install error

schindlershadow, 2015-04-05:
This may be an npm issue, but i'm not sure.

Getting the following error when installing:

$ npm install msi-keyboard
npm WARN package.json led-manager@ No description
npm WARN package.json led-manager@ No repository field.
npm WARN package.json led-manager@ No README data
|
> node-hid@0.3.2 preinstall /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules/node-hid
> sh get-hidapi.sh

Cloning into 'hidapi'...
remote: Counting objects: 1965, done.
remote: Total 1965 (delta 0), reused 0 (delta 0), pack-reused 1965
Receiving objects: 100% (1965/1965), 2.79 MiB | 1.06 MiB/s, done.
Resolving deltas: 100% (1082/1082), done.
Checking connectivity... done.

> node-hid@0.3.2 install /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules/node-hid
> sh install.sh

make: Entering directory '/home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
../hidapi/libusb/hid.c: In function ‘hid_read_timeout’:
../hidapi/libusb/hid.c:1078:6: warning: variable ‘bytes_read’ might be clobbered by ‘longjmp’ or ‘vfork’ [-Wclobbered]
  int bytes_read = -1;
      ^
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
../src/HID.cc: In member function ‘virtual v8::Handle<v8::Value> JSException::asV8Exception() const’:
../src/HID.cc:50:71: error: ‘New’ is not a member of ‘v8::String’
   virtual Handle<Value> asV8Exception() const { return ThrowException(String::New(message().c_str())); }
                                                                       ^
../src/HID.cc:50:101: error: ‘ThrowException’ was not declared in this scope
   virtual Handle<Value> asV8Exception() const { return ThrowException(String::New(message().c_str())); }
                                                                                                     ^
../src/HID.cc: At global scope:
../src/HID.cc:58:1: error: expected class-name before ‘{’ token
 {
 ^
../src/HID.cc:61:38: error: ‘Arguments’ does not name a type
   static Handle<Value> devices(const Arguments& args);
                                      ^
../src/HID.cc:76:34: error: ‘Arguments’ does not name a type
   static Handle<Value> New(const Arguments& args);
                                  ^
../src/HID.cc:77:35: error: ‘Arguments’ does not name a type
   static Handle<Value> read(const Arguments& args);
                                   ^
../src/HID.cc:78:36: error: ‘Arguments’ does not name a type
   static Handle<Value> write(const Arguments& args);
                                    ^
../src/HID.cc:79:36: error: ‘Arguments’ does not name a type
   static Handle<Value> close(const Arguments& args);
                                    ^
../src/HID.cc:80:45: error: ‘Arguments’ does not name a type
   static Handle<Value> setNonBlocking(const Arguments& args);
                                             ^
../src/HID.cc:81:47: error: ‘Arguments’ does not name a type
   static Handle<Value> getFeatureReport(const Arguments& args);
                                               ^
../src/HID.cc:83:48: error: ‘Arguments’ does not name a type
   static Handle<Value> sendFeatureReport(const Arguments& args);
                                                ^
../src/HID.cc:86:25: error: ‘uv_work_t’ has not been declared
   static void recvAsync(uv_work_t* req);
                         ^
../src/HID.cc:87:29: error: ‘uv_work_t’ has not been declared
[... 364 lines cut ...]
158 verbose exit [ 1, true ]
159 verbose unbuild node_modules/msi-keyboard/node_modules/node-hid
160 info preuninstall node-hid@0.3.2
161 info uninstall node-hid@0.3.2
162 verbose unbuild rmStuff node-hid@0.3.2 from /home/schindler/Downloads/msi-klm/node_modules
163 verbose unbuild rmStuff in /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules
164 info postuninstall node-hid@0.3.2
165 silly gentlyRm /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules/node-hid is being purged from base /home/schindler/Downloads/msi-klm
166 verbose gentlyRm don't care about contents; nuking /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules/node-hid
167 silly vacuum-fs purging /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules/node-hid
168 silly vacuum-fs removing /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard/node_modules
169 silly vacuum-fs quitting because other entries in /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard
170 verbose unbuild node_modules/msi-keyboard
171 info preuninstall msi-keyboard@0.0.2
172 info uninstall msi-keyboard@0.0.2
173 verbose unbuild rmStuff msi-keyboard@0.0.2 from /home/schindler/Downloads/msi-klm/node_modules
174 info postuninstall msi-keyboard@0.0.2
175 silly gentlyRm /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard is being purged from base /home/schindler/Downloads/msi-klm
176 verbose gentlyRm don't care about contents; nuking /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard
177 silly vacuum-fs purging /home/schindler/Downloads/msi-klm/node_modules/msi-keyboard
178 silly vacuum-fs removing /home/schindler/Downloads/msi-klm/node_modules
179 silly vacuum-fs finished vacuuming up to /home/schindler/Downloads/msi-klm

$ node --version
v0.12.2

stevelacy, 2015-04-06:
https://github.com/node-hid/node-hid/issues/90

closed 2015-04-06 by stevelacy

phaazon, 2015-04-06:
Ok so you close the issue “until node-hid fixes it”?

stevelacy, 2015-04-06:
@phaazon what issue?

phaazon, 2015-04-07:
This very one. I don’t get what the message is then. Your tool is broken because of a dependency. Does it mean you won’t provide any patch until the dependency updates and gets fixed?

stevelacy, 2015-04-08:
This relies on a node module called `node-hid`, it can not be 'patched' to work on v0.12 as it relies on `node-hid` to control the HID devices.
#13 is currently tracking the progress of `node-hid` and v0.12