stephenlacy/msi-keyboard #23: MSI GE70 issues

makkarpov, 2015-05-02:
I tested this code on MSI GE-70 in a static color mode and found two issues:

1) The 'green' color is not actually green. It looks more like yellow. Also, original control software do not limits user to a set of colors, it allows selection from full RGB space. Maybe there is a protocol that accepts RGB values?

2) Hardware backlight button acts like hardware backlight switch. But original driver hooks this button and change backlight layers on press.

Unfortunately, at this time I cannot reverse a keyboard protocol from original Windows drivers. I cannot get these drivers working in VirtualBox.

stevelacy, 2015-05-03:
There is no 'protocol', it is just a HID device which accepts numbers as colors.
The windows software has a key-mapping from the RGB to number values.
Refer to this list for found colors: https://github.com/wearefractal/msi-keyboard/issues/11#issuecomment-72904478

The hardware button is dependent on the version of the keyboard and the MSI motherboard controls. My GS-60 ghost pro only has illumination levels from off to high. My MSI GT-60 had the modes regardless of OS.

Since this is a node module for accessing the interface and allowing customization I would not add the RGB conversion to it directly, but rather to a GUI project.

Would you be interested in creating an [electron](https://github.com/atom/electron) GUI app for the keyboard? I have been planning to do so, but have not had the time.

makkarpov, 2015-05-03:
When I refered to a 'protocol', I refered to a format of HID reports. Thanks for a list of colors, so some facts that I found on a my laptop:

1) Green is a color 10, color 4 looks more like yellow on my laptop. Maybe this is keyboard-dependent.

[image]
[image]

2) Levels are inverted, so 'high' (0 as defined in constants.json) is most darkest level, and 'light' (3) is most bright.

3) Red (1) at brightness 1 is red and red (1) at brightness 3 is pink, but red (9) is same color with different brightness. Again, this may be keyboard-dependent.

I will try to reverse color alghorithm from original Steelseries software, but there is no guarantee that I will succeed in it

closed 2016-05-05 by stevelacy