stephenlacy/msi-keyboard #24: Configure Error when trying to install

Aidoboy, 2015-06-04:
I used this on Ubuntu 14.10 and loved it, but I accidentally deleted my home folder and decided to install 15.04 fresh. This is one of the things I'm trying to reinstall, but it doesn't want to install. This is the error I'm getting:
```
$ sudo npm install

> node-hid@0.4.0 install /home/aidan/Code/MSI-Backlight/msi-keyboard/node_modules/node-hid
> node-gyp rebuild

/bin/sh: 1: node: not found
gyp: Call to 'node -e "require('nan')"' returned exit status 127. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:344:16)
gyp ERR! stack     at ChildProcess.EventEmitter.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:797:12)
gyp ERR! System Linux 3.19.0-18-generic
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/aidan/Code/MSI-Backlight/msi-keyboard/node_modules/node-hid
gyp ERR! node -v v0.10.25
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok 
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian
 
npm ERR! node-hid@0.4.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.4.0 install script.
npm ERR! This is most likely a problem with the node-hid package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls node-hid
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 3.19.0-18-generic
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install"
npm ERR! cwd /home/aidan/Code/MSI-Backlight/msi-keyboard
npm ERR! node -v v0.10.25
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! 
npm ERR! Additional logging details can be found in:
npm ERR!     /home/aidan/Code/MSI-Backlight/msi-keyboard/npm-debug.log
npm ERR! not ok code 0
```
Any help? Thanks in advance if so!

stevelacy, 2015-06-04:
`/bin/sh: 1: node: not found`

Aidoboy, 2015-06-08:
How do I do that?

jamengual, 2015-06-09:
sudo ln -s /usr/bin/nodejs  /usr/bin/node

Aidoboy, 2015-06-10:
I think that worked, but now this happens when I try to run an example:
`aidan@cinnabar:~/Source/msi-keyboard/examples$ sudo nodejs msi-kbd.js --help

module.js:340
    throw err;
          ^
Error: Cannot find module 'msi-keyboard'
    at Function.Module._resolveFilename (module.js:338:15)
    at Function.Module._load (module.js:280:25)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at Object.<anonymous> (/home/aidan/Source/msi-keyboard/examples/msi-kbd.js:7:16)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)`

stevelacy, 2015-06-11:
```
Error: Cannot find module 'msi-keyboard'
```

If you see this issue the easy fix is to run this from the file's directory:

`npm install msi-keyboard`

Or, if you do not want to install a second time, edit the js example file and change the:

```
require('msi-keyboard');
```
To:

```
require('../');  // this requires the 'parent' 
```

And in the parent directory (the root of the project) run `npm install`.

The examples are 'stand alone', thus the need to install or modify. 

If you have any issues please ask!

Aidoboy, 2015-06-11:
Thanks a lot!

closed 2015-06-11 by Aidoboy

hyperqwestnetwork, 2015-08-22:
Ok,  I am a noob when it comes to this, Could you point me in the right direction on how to use this program to control my keyboard.

Thank you