stephenlacy/msi-keyboard #25: Segfault on GS60

bparker06, 2015-08-16:
I have the gs60 pro 4k with steelseries keyboard and I'm trying to change the mode, but the program segfaults:

Core was generated by `node msi-run.js'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  instance_type (this=<error reading variable: Cannot access memory at address 0x0>) at ../src/objects-inl.h:3007
3007	../src/objects-inl.h: No such file or directory.
(gdb) bt
#0  instance_type (this=<error reading variable: Cannot access memory at address 0x0>) at ../src/objects-inl.h:3007
#1  IsString (this=0x1) at ../src/objects-inl.h:177
#2  VerifyApiCallResultType (this=0x1) at ../src/objects-inl.h:1718
#3  HandleApiCallHelper<true> (isolate=0x19fe070, args=...) at ../src/builtins.cc:1152
#4  v8::internal::Builtin_HandleApiCallConstruct (args=..., isolate=0x19fe070) at ../src/builtins.cc:1169
#5  0x0000032290206362 in ?? ()
#6  0x0000000001a5dfd0 in ?? ()
#7  0x00000322902062c1 in ?? ()
#8  0x00007fff3e30a4e0 in ?? ()
#9  0x00007fff3e30a550 in ?? ()
#10 0x000003229020d4f6 in ?? ()
#11 0x0000281ef0658af9 in ?? ()
#12 0x0000ff0000000000 in ?? ()
#13 0x0000177000000000 in ?? ()
#14 0x000032be7df5c979 in ?? ()
#15 0x000032be7df5c979 in ?? ()
#16 0x0000000200000000 in ?? ()
#17 0x000003229020d321 in ?? ()
#18 0x0000000700000000 in ?? ()
#19 0x000032be7df0d009 in ?? ()
#20 0x00007fff3e30a598 in ?? ()
#21 0x000003229020d7aa in ?? ()
#22 0x0000ff0000000000 in ?? ()
#23 0x0000177000000000 in ?? ()
#24 0x0000281ef0607c19 in ?? ()
#25 0x0000281ef0658af9 in ?? ()
#26 0x000003229020d701 in ?? ()
#27 0x0000000600000000 in ?? ()
#28 0x0000000000000000 in ?? ()

Any idea what's going on? This is the contents of the file I'm running:
```
var keyboard = require('msi-keyboard');

keyboard.mode('normal');
```

stevelacy, 2015-08-16:
Looks like the module was not built correctly:  `../src/objects-inl.h: No such file or directory.`

What is your node version?
What is the output from the npm build?

bparker06, 2015-08-17:
$ node -v
v0.10.25

$ npm install
npm http GET https://registry.npmjs.org/node-hid
npm http GET https://registry.npmjs.org/should
npm http GET https://registry.npmjs.org/mocha
npm http 304 https://registry.npmjs.org/node-hid
npm http 304 https://registry.npmjs.org/should
npm http 304 https://registry.npmjs.org/mocha
npm http GET https://registry.npmjs.org/should-equal/0.5.0
npm http GET https://registry.npmjs.org/should-format/0.3.0
npm http GET https://registry.npmjs.org/should-type/0.2.0
npm http GET https://registry.npmjs.org/commander/2.3.0
npm http GET https://registry.npmjs.org/debug/2.0.0
npm http GET https://registry.npmjs.org/diff/1.4.0
npm http GET https://registry.npmjs.org/escape-string-regexp/1.0.2
npm http GET https://registry.npmjs.org/glob/3.2.3
npm http GET https://registry.npmjs.org/growl/1.8.1
npm http GET https://registry.npmjs.org/jade/0.26.3
npm http GET https://registry.npmjs.org/mkdirp/0.5.0
npm http GET https://registry.npmjs.org/supports-color
npm http GET https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/commander/2.3.0
npm http 304 https://registry.npmjs.org/should-equal/0.5.0
npm http 304 https://registry.npmjs.org/diff/1.4.0
npm http 304 https://registry.npmjs.org/escape-string-regexp/1.0.2
npm http 304 https://registry.npmjs.org/debug/2.0.0
npm http 304 https://registry.npmjs.org/glob/3.2.3
npm http 304 https://registry.npmjs.org/should-format/0.3.0
npm http 304 https://registry.npmjs.org/mkdirp/0.5.0
npm http 304 https://registry.npmjs.org/jade/0.26.3
npm http 304 https://registry.npmjs.org/supports-color
npm http 304 https://registry.npmjs.org/nan
npm http 304 https://registry.npmjs.org/growl/1.8.1
npm http GET https://registry.npmjs.org/minimist/0.0.8
npm http GET https://registry.npmjs.org/ms/0.6.2
npm http GET https://registry.npmjs.org/minimatch
npm http GET https://registry.npmjs.org/graceful-fs
npm http GET https://registry.npmjs.org/inherits
npm http GET https://registry.npmjs.org/commander/0.6.1
npm http GET https://registry.npmjs.org/mkdirp/0.3.0

> node-hid@0.4.0 install /home/bp/msi-keyboard/node_modules/node-hid
> node-gyp rebuild

npm http 304 https://registry.npmjs.org/minimist/0.0.8
npm http 304 https://registry.npmjs.org/graceful-fs
npm http 304 https://registry.npmjs.org/minimatch
npm http 304 https://registry.npmjs.org/inherits
npm http 200 https://registry.npmjs.org/commander/0.6.1
npm http 304 https://registry.npmjs.org/mkdirp/0.3.0
npm http GET https://registry.npmjs.org/commander/-/commander-0.6.1.tgz
make: Entering directory `/home/bp/msi-keyboard/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
npm http 304 https://registry.npmjs.org/ms/0.6.2
npm http GET https://registry.npmjs.org/sigmund
npm http GET https://registry.npmjs.org/lru-cache
npm http 200 https://registry.npmjs.org/commander/-/commander-0.6.1.tgz
npm http 304 https://registry.npmjs.org/sigmund
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
npm http 200 https://registry.npmjs.org/lru-cache
../src/HID.cc: In static member function ‘static v8::Handle<v8::Value> HID::New(const v8::Arguments&)’:
../src/HID.cc:363:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../src/HID.cc: In static member function ‘static v8::Handle<v8::Value> HID::close(const v8::Arguments&)’:
../src/HID.cc:377:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../src/HID.cc: In static member function ‘static v8::Handle<v8::Value> HID::setNonBlocking(const v8::Arguments&)’:
../src/HID.cc:396:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
../src/HID.cc: In static member function ‘static v8::Handle<v8::Value> HID::write(const v8::Arguments&)’:
../src/HID.cc:424:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
  SOLINK_MODULE(target) Release/obj.target/HID.node
  SOLINK_MODULE(target) Release/obj.target/HID.node: Finished
  COPY Release/HID.node
make: Leaving directory `/home/bp/msi-keyboard/node_modules/node-hid/build'
npm http 304 https://registry.npmjs.org/should-type/0.2.0
mocha@2.2.5 node_modules/mocha
├── escape-string-regexp@1.0.2
├── supports-color@1.2.1
├── growl@1.8.1
├── commander@2.3.0
├── diff@1.4.0
├── mkdirp@0.5.0 (minimist@0.0.8)
├── debug@2.0.0 (ms@0.6.2)
├── jade@0.26.3 (commander@0.6.1, mkdirp@0.3.0)
└── glob@3.2.3 (inherits@2.0.1, graceful-fs@2.0.3, minimatch@0.2.14)

node-hid@0.4.0 node_modules/node-hid
└── nan@1.9.0

should@7.0.4 node_modules/should
├── should-type@0.2.0
├── should-equal@0.5.0
└── should-format@0.3.0
$ node
> var keyboard = require('msi-keyboard');
Segmentation fault (core dumped)

stevelacy, 2015-08-19:
Do you have build-essential installed?
Try node v10.38 or iojs 2.5

My setup:
uname: 3.13.0-37-generic
iojs: v2.5.0

Same ghost pro as yours

bparker06, 2015-08-23:
I didn't want to install versions other than the system one, I ended up just writing a C program that does the same thing using hidapi/libusb.

madscientist42, 2015-09-05:
Still segfaults on the stock node.js from Ubuntu.  It should, to be blunt, work on any of the engines for JS instead of the one you're specifically on.

giuliopaci, 2015-09-08:
@bparker06: are you going to share the code for that application? I would prefer a C application if it exists.

stevelacy, 2015-09-09:
@madscientist42 To be blunt. This is a OSS module. I do not need 
to support any particular node version YOU are on.

Part of the debug process is to test the version against other kernels, package versions and node versions. Just commenting a:
> Still segfaults on the stock node.js from Ubuntu.

Without any additional information is not in anyway helpful.
This is the first issue where a segfault was listed. ALL segfaults in this module are related to `node-hid` and or it's dependencies as that interfaces with the C++ libs directly.

---

@bparker06 Thanks for reporting, feel free to contact me if you noticed any possible solutions.

@giuliopaci @bparker06  I would also like to see it. I will lock this issue as it is going away from the initial topic.

closed 2015-09-09 by stevelacy