stephenlacy/msi-keyboard #26: C++ port of msi-keyboard

bparker06, 2015-09-11:
I could no longer comment on issue #25 as it was closed, but here is the code that @stevelacy and @giuliopaci asked for:

https://github.com/bparker06/msi-keyboard

Not trying to step on any toes here; if it's not appropriate I'll gladly take it down.

stevelacy, 2015-09-11:
Thanks, I'll check it out

giuliopaci, 2015-09-14:
Thank you very much. I will also try it soon. :-)

timofonic, 2016-01-28:
Thanks!

Could this be integrated into Linux and BSDs in a standardized way? By using proc or any other interface

giuliopaci, 2016-01-28:
@timofonic, what do you mean by "standardized" way?
@bparker06, I just noticed that I forgot to update you about the fact that I am an happy user of this application now. I created a Debian package for it, whose sources are here on github.com. I am also thinking about trying to have it included in the official distribution (it requires time).

timofonic, 2016-01-28:
@giuliopaci 

I mean a standarized HID access to this feature, available on other devices too.

By exposing it in a popular interface, more software could make use of it for more creative uses ;)

timofonic, 2016-09-15:
@giuliopaci 

There are some suport in Linux kernel about HID and lights. What about contacting MSI about this?

Any ideas?

closed 2022-05-02 by stevelacy