stephenlacy/msi-keyboard #28: Error: could not send feature report to device

folx, 2015-10-11:
Hi there,

Trying msi-keyboard on a fresh (brand new GS60 2PL) Funtoo install got me following stack trace:
.../msi-keyboard/examples $ node sample.js 
.../msi-keyboard/lib/setColor.js:24
	keyboard.sendFeatureReport(activate);
	         ^

Error: could not send feature report to device
    at Error (native)
    at module.exports (.../msi-keyboard/lib/setColor.js:24:11)
    at HID.board.color (.../msi-keyboard/lib/findKeyboard.js:12:3)
    at Object.<anonymous> (.../msi-keyboard/examples/sample.js:3:10)
    at Module._compile (module.js:434:26)
    at Object.Module._extensions..js (module.js:452:10)
    at Module.load (module.js:355:32)
    at Function.Module._load (module.js:310:12)
    at Function.Module.runMain (module.js:475:10)
    at startup (node.js:117:18)

Currently using nodejs-4.1.1 & libusb-1.0.19, is there a dependency I missed?

stevelacy, 2015-10-12:
Confirmed, the error is found on node v4.1.1

Fresh install Debian 8, Linux kernel 4.2.0.
node: 4.1.1 - error
iojs: 2.5.0 - no error

I attempted to determine if the error was fixed in a recent version of nan, not so apparently.
Want to open an issue on `node-hid` as that is the interface used?

folx, 2015-10-18:
In deed nodejs 0.12.6 & 0.12.7 work just fine (thus haven't tryed iojs); Thanks for the tip.

So I submited the very same report to node-hid: https://github.com/node-hid/node-hid/issues/118

closed 2015-10-18 by folx

JuanCAlpizar, 2015-10-28:
I'm having this same issue but using node 4.2.1, how can I update (?) to nodejs 0.12.6 and how do they differ?

hash-bang, 2015-11-09:
Same problem here using node 4.2.2

Any help would be appreciated

hash-bang, 2015-11-12:
Found a work around for this using slightly modified instructions from @folx.

I think the issues with @folx example is that a) node-hid needs to be compiled specifically for the active node version and b) the node file needs to be run as root to have permissions to access the HID layer.

1. Install Nave: `sudo npm install -g nave`
2. Create a project directory: `mkdir msi-keyboard`
3. ... and enter it: `cd msi-keyboard`
4. Boot Nave and have it install the NPM: `npm install msi-keyboard`. This is important as `node-hid` needs to be rebuilt to the exact node version
5. Add a simple `app.js` file:

    	var keyboard = require('msi-keyboard');

	keyboard
		.color('left', 'red')
		.color('middle', {color: 'red', intensity: 'med'})
		.color('right', {color: 'red', intensity: 'med'})

6. Exit the nave shell: `exit`
7. To run the file use `sudo nave use 0.12.6 node app.js`