stephenlacy/msi-keyboard #30: Cannot find module

RD1945, 2015-11-19:
Hello,

After I installed the necessary modules by typing 'npm install' and run an example from the program "nodejs examples/blink.js" I get the following error: Cannot find module '/home/michel/Downloads/.msi-keyboard/node_modules/node-hid/build/HID/v0.5.1/Release/node-v11-linux-x64/HID.node'.

If I check the folders there is no node-v11-linux-x64. Only a node-v47-linux-x64

Please help

stevelacy, 2015-11-19:
What is your node version?
It is an error with `node-hid`

RD1945, 2015-11-19:
Node version: 4.2.2
Nodejs version: 0.10.25

Tried with only node installed, no luck.
Tried with only nodejs installed, no luck.
Tried with both installed, no luck.

All the dependencies are met (I think).

stevelacy, 2015-11-19:
Until `node-hid` fixes: https://github.com/node-hid/node-hid/issues/118 use node v12 or iojs v2.5.0

closed 2016-02-08 by stevelacy