stephenlacy/msi-keyboard #34: Origin evo15-s

iAL-Ahmed, 2016-01-09:
I have an Origin evo15-s, which to my knowledge shares the same keyboard with the msi gs60, but none of the example scripts or my own seem to work. It sits on the default red color and doesn't change.

I ran

```
var hid = require('node-hid');
console.log(hid.devices());

var board = new hid.HID(6000, 65280);
console.log(board);
```
and got 

```
[ { vendorId: 6000,
    productId: 65280,
    path: '0004:0003:00',
    serialNumber: 'MSI EPF USB',
    manufacturer: 'MSI EPF USB',
    product: 'MSI EPF USB',
    release: 272,
    interface: 0 }]

{ domain: null,
  _events: { newListener: [Function] },
  _maxListeners: 10,
  _raw: {},
  read: [Function],
  write: [Function],
  getFeatureReport: [Function],
  sendFeatureReport: [Function],
  setNonBlocking: [Function],
  _paused: true }
```
So it appears to indeed be an MSI board. 

My intent is to make up for Origin pc's lack of keyboard control, even though the hardware should be customizable. 

Any thoughts?

Thanks.

stevelacy, 2016-01-11:
What node version, any error?

iAL-Ahmed, 2016-01-12:
Thanks for the reply! I am running version 5.4.0 and I am getting this error:

```
Error: could not send feature report to device
    at Error (native)
    at module.exports (/home/issaq/Desktop/msi stuff/msi-keyboard/lib/setColor.js:24:11)
    at HID.board.color (/home/issaq/Desktop/msi stuff/msi-keyboard/lib/findKeyboard.js:12:3)
    at Object.<anonymous> (/home/issaq/Desktop/msi stuff/msi-keyboard/examples/blink.js:4:10)
    at Module._compile (module.js:398:26)
    at Object.Module._extensions..js (module.js:405:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Function.Module.runMain (module.js:430:10)
    at startup (node.js:141:18)
```

stevelacy, 2016-01-12:
did you try using sudo?

iAL-Ahmed, 2016-01-12:
Yeah.

```
sudo node blink
```

stevelacy, 2016-02-08:
Related to this: https://github.com/node-hid/node-hid/issues/118
I will check to see if they fixed the issues with node v2.5+

stevelacy, 2016-05-05:
The newest release should have fixed this issue :)
Please open if the issue persists.

closed 2016-05-05 by stevelacy