stephenlacy/msi-keyboard #37: Could not send feature report to device

dimitriscsd, 2016-03-11:
Hello.

I am using the CLI version installed through npm. It has a dependency on this library, which I was using in the past (as standalone) with great success.

I recently had to update my node version for work and I'm now using 5.8.0

Unfortunately it has stopped working and when I try "sudo msikeyboard -c green" I get the following error:

/usr/local/lib/node_modules/msi-keyboard-cli/node_modules/msi-keyboard/lib/setColor.js:24
	keyboard.sendFeatureReport(activate);
	         ^

Error: could not send feature report to device
    at Error (native)
    at module.exports (/usr/local/lib/node_modules/msi-keyboard-cli/node_modules/msi-keyboard/lib/setColor.js:24:11)
    at HID.board.color (/usr/local/lib/node_modules/msi-keyboard-cli/node_modules/msi-keyboard/lib/findKeyboard.js:12:3)
    at /usr/local/lib/node_modules/msi-keyboard-cli/msi-keyboard-CLI.js:348:18
    at FSReqWrap.cb [as oncomplete] (fs.js:216:19)

dimitriscsd, 2016-03-11:
I should have checked previous issues before I posted this. Sorry.

It looks like this is a node-hid issue. Not yours.

Assuming that they one day fix this on their end, will an update of msi-keyboard-cli be enough to get it to work? I guess you'd have to reference the newer version of node-hid as a dependency?

stevelacy, 2016-03-11:
Yes, assuming that they fix the issue I will update and publish this module with the newer `node-hid` dependency. Unfortunately I also need to use iojs 2.5 with `node-hid`

PR #40 (2016-05-02): sendFeatureReport with regular array instead of node Buffer
Ref: https://github.com/node-hid/node-hid/issues/122
Ref: https://github.com/node-hid/node-hid/issues/118
Fixes: https://github.com/Kwaadpepper/msi-keyboard-CLI/issues/6
Fixes: #37

This is a temporary work around until node-hid can handle Buffers. ArrayBuffers seem to work under newer node versions (tested on 5.4.1), but they still cause the same issue with older iojs versions (2.5.0).
closed 2016-05-02 by stevelacy