stephenlacy/msi-keyboard #39: How did you find info for constants?

visioninit, 2016-04-18:
https://github.com/stevelacy/msi-keyboard/blob/master/lib/constants.json

I'm trying to figure out if someone could use the info to create a driver that doesn't require sudo. I haven't been able to find the technical information that describes anything. Guessing either MSI or steelseries has the full info but how do I explore the interface at a lower level?

stevelacy, 2016-04-18:
https://github.com/stevelacy/msi-keyboard/issues/11#issuecomment-72799841
Reverse engineering of the older steelseries application.

The hid controller requires sudo, to remove the need for sudo would mean giving your unix user permissions over the device. https://www.tincantools.com/wiki/Accessing_Devices_without_Sudo

This is the smallest C++ code I could write to control the keyboard:
```c
#include <iostream>
#include <hidapi/hidapi.h>

#define BUFSIZE 8
int main() {
  int mode = 3;
  // 6 = breathe with pre-set colors
  // 2 = game mode
  unsigned char buf[BUFSIZE] = {0};
  buf[0] = 1;
  buf[1] = 2;
  buf[2] = 65;
  buf[3] = mode;
  buf[4] = 0;
  buf[5] = 0;
  buf[6] = 0;
  buf[7] = 236;

  hid_device *keyboard = hid_open(6000, 65280, 0);
  hid_send_feature_report(keyboard, buf, BUFSIZE);

}

```

visioninit, 2016-04-19:
Very cool, thanks! That's awesome of you to share that. I'll play around with this (first I'm going to have to figure out how to get it running, I have the ubuntu SDK installed that supports C, but I'm a noob.)  The udev rule thing I've seen before with connecting phones with the android SDK so that makes sense and I don't see any issues with adding that rule.

closed 2016-04-19 by visioninit