stephenlacy/msi-keyboard #42: MSI GT62-2QE Error: cannot open device with vendor id 0x1770 and product id 0xff00

Ekdohibs, 2016-05-19:
I know this error is usually caused by not running at root, however I *do* run the server as root.
Moreover, the device isn't detected in lsusb, even when running it as root. Any idea of what the problem might be?

Ekdohibs, 2016-05-19:
Sorry for incorrectly reporting this error: I thought my laptop supported keyboard backlight while it doesn't.

closed 2016-05-19 by Ekdohibs

stevelacy, 2016-05-19:
Ah, does it do anything on windows?

Ekdohibs, 2016-05-19:
Sorry, can't try, the laptop was shipped without an OS. The manual shipped with the laptop mentioned a keyboard shortcut for changing the keyboard brightness level, which led me to think it had keyboard backlight; however, the laptop description on MSI's website does not talk about keyboard backlight.

stevelacy, 2016-05-19:
Ah, gotcha