stephenlacy/msi-keyboard #43: Ability to set arbitrary RGB colors

makkarpov, 2016-06-06:
On my notebook (GE70) Steelseries keyboard seems to support plain RGB values which are much simplier to specify than color-intensity pairs. All packet dumps are specified in hex:

1. Normal mode: 
   1. `01 02 41 01 00 00 00 00`
   2. `01 02 40 01 RR GG BB 00` — color for left region
   3. `01 02 40 02 RR GG BB 00` — color for middle region
   4. `01 02 40 03 RR GG BB 00` — color for right region

3. Breathing/Wave/Dual color:
   1. `01 02 44 01 RR GG BB 00` — left, primary
   2. `01 02 44 02 RR GG BB 00` — left, secondary
   3. `01 02 44 03 TT TT TT 00` — timings
   4. `01 02 41 MM 00 00 00 00` — where `MM` is `03` for breathing, `05` for wave, `06` for dual color
   5. repeat two more times with fourth byte set to `04`, `05`, `06` for middle region and `07`, `08`, `09` for right region.

Full dumps are [here](http://pastebin.com/kkKRgBmW). I already implemented code based on these dumps in C++ and it seems to work well.

stevelacy, 2016-06-06:
PR?

makkarpov, 2016-06-06:
AFAIK, currently it requires almost full architecture change - too many parts are tied to color-intensity pairs. Since I know node.js very little, I'm afraid that it's better to leave code for someone that is more experienced with node.js

Also I don't know which hardware supports these packets, so some testing will be appreciated.

stevelacy, 2016-06-06:
Ok, could you upload your c++ to gh? I could then port it to js

Thanks

makkarpov, 2016-06-06:
Here is it: https://gist.github.com/makkarpov/b559a82724158adeecf195e26ef3d951

JuanCAlpizar, 2017-01-02:
I also don't know much about NodeJS but I do know about JavaScript, HTML 5 and CSS 3. This is a bit off topic because what I made was the support for another keyboard but it also has a rebuild of the color interface to support RGB based color combinations, please check out issue #20 on the matter

stevelacy, 2017-01-03:
I'd say talk about the colors here, and the regions on a new issue.
Node is just like a browser's JS engine, same language, slightly different interface.

JuanCAlpizar, 2017-01-04:
Where should I begin? I did fork the most recent msi-keyboard master branch and made the modifications parting from that code. I did it on a separate branch on my own branch, so you're free to check it out on https://github.com/JuanCAlpizar/msi-keyboard/commit/f169552ab7c41b3b42527b8363cc459015e940ea

- I didn't modify the amount of parameters or function names, so it should be compatible backwards, maybe a little extra validation on additional regions for the keyboards that don't have these regions would do the thing.

- The main change for it it's on the `constants.json` file. As far as I understand the colors you were sending are preset colors by the keyboard. In order to send custom colors you need to send RGB values, so instead of a a single id color bound to such color like this

     "colors": {
		"black": 0,
		"red": 1,
                [...],
      },"

I changed the json constants for colors in objects with red, green and blue values, like this:

     "colors": {
	"black": {
		"redScale": 0,
		"greenScale": 0,
		"blueScale": 0
	},
	"red": {
		"redScale": 255,
		"greenScale": 0,
		"blueScale": 0
	},
	[...],
     }"

The next step is to change the parameters to send in the array to `sendData` and `sendFeatureRequest` methods. For simple color set, like the one on `setColor.js` you need to change the mode from 66 to 64, this mode sends red, green and blue values as 3 separate digits on the 4, 5 and 6 indexes of the array. This leaves no space left to send intensity levels (assuming `sendFeatureRequest` actually sends a byte of information, hence the 8 indexes on the array sent) so I changed intensity to be a percentage of the red, green and blue sent values, making it dimmer or brighter, but still the same color; if you want to keep the color degradation to white (current "intensity" on your code, didn't notice until I was done) I think you could still get the same effect but the intensity multiplier definition would be different. Here's a sample of the array sent:

 	activate[0] = 1;
	activate[1] = 2;
	activate[2] = 64; // set color
	activate[3] = constants.regions[region];
	activate[4] = constants.colors[color].redScale * constants.levels[intensity];
	activate[5] = constants.colors[color].greenScale * constants.levels[intensity];
	activate[6] = constants.colors[color].blueScale * constants.levels[intensity];
	activate[7] = 236;  // EOR (end of request)

For breathe and wave mode you has a slight modification on the array sent with `sendData`. Each region still has 3 layers: one for foreground color, one for background color and one for the execution period of the region. To send the colors the mode has to change from 67 to 68, which allows to send to colors with rgb parameters and cycle through them. The execution time layer still needs to be sent using the 67 mode. An example of the color arrays for breathe and wave modes is as follows;

 	data[0] = 1;
	data[1] = 2;
	data[2] = 68;
	data[3] = val1;
	data[4] = val2.redScale;
	data[5] = val2.greenScale;
	data[6] = val2.blueScale;
	data[7] = 236;

stevelacy, 2017-01-04:
We should probably break backwards compatibility if we add support for RGB. This would let the gui app use a color picker with a full spectrum for the colors.
The module in general should be rebuilt to clean up the functions and document the features, I'll tackle that and then add your commit for RGB colors.

Thanks!

stellarpower, 2017-09-23:
I will try and have a look at this as soon as I can. Out of interest, what packet dumping utility did you use?

On another note, I imagine the newer devices with per-key colours use a newer controller but I would love to experiment and see if the steelseries in my laptop would respond to the commands for this. If anyone has aa dump I'd be really grateful to play with it.

Thanks

stevelacy, 2017-10-17:
Thanks to @stellarpower we now have full https://github.com/Marak/colors.js colors!

closed 2017-10-17 by stevelacy