stephenlacy/msi-keyboard #44: new code breaks msi keyboard

bash64, 2016-06-30:
I own an MSI-GT70-2PE laptop.

Here is the code I use to compile for linux mint 17 (ubuntu 14.04):
>sudo apt-get install git libusb-dev libusb-1.0-0-dev nodejs npm build-essential --install-recommends
>sudo ln -s /usr/bin/nodejs /usr/bin/node
>git clone https://github.com/wearefractal/msi-keyboard
>git clone https://github.com/hanshuebner/node-hid
>cd ~/node-hid
>git clone https://github.com/signal11/hidapi.git
>cd ~/node-hid; sudo npm install -g
>cd ~/msi-keyboard; sudo npm install --g

Until recently it worked. Then I reinstalled using the latest github code and it broke.
Some sort of change to the code has caused it to quit working.
I had a copy of the old github code and recovered it and it compiled and worked fine.
Thank God for backups.

I am now upgraded to Mint 18 (ubuntu 16.04):

Here is the error in Mint 18 with the old github code that works in Mint 17:
**>nodejs rainbow.js 
/usr/local/lib/node_modules/msi-keyboard/node_modules/node-hid/nodehid.js:28
	this._raw = new (Function.prototype.bind.apply(binding.HID,
	            ^

Error: cannot open device with vendor id 0x1770 and product id 0xff00
    at Error (native)
    at new HID (/usr/local/lib/node_modules/msi-keyboard/node_modules/node-hid/nodehid.js:28:14)
    at module.exports (/usr/local/lib/node_modules/msi-keyboard/lib/findKeyboard.js:7:14)
    at Object.<anonymous> (/usr/local/lib/node_modules/msi-keyboard/index.js:3:18)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Module.require (module.js:354:17)
    at require (internal/module.js:12:17)**

Here is the error in Mint 18 with the new github code:
**>nodejs rainbow.js 
/home/dad/rainbow.js:2
keyboard.color('left',{color: 'red', intensity: 'med'});
         ^

TypeError: keyboard.color is not a function
    at Object.<anonymous> (/home/dad/rainbow.js:2:10)
    at Module._compile (module.js:410:26)
    at Object.Module._extensions..js (module.js:417:10)
    at Module.load (module.js:344:32)
    at Function.Module._load (module.js:301:12)
    at Function.Module.runMain (module.js:442:10)
    at startup (node.js:136:18)
    at node.js:966:3**

All help appreciated.

If you need more info just ask.

[rainbow.js.zip](https://github.com/stevelacy/msi-keyboard/files/340700/rainbow.js.zip)

bash64, 2016-06-30:
Here is a copy of the old github code that worked:
https://mega.nz/#!WRoBBLQC!okyzWQSGUIw7A4LpuDCyS05VVxquQSaty8S5YDdhgVI

stevelacy, 2016-06-30:
git, as well as github do have a complete history of the changes:
https://github.com/stevelacy/msi-keyboard/commits/master

It sounds like the error is when you updated your OS.
The `node-hid` module _needs_ to be compiled for each kernel and node version. Please add your node version as that will help. (node -v)

Try this to start:

```
git clone https://github.com/stevelacy/msi-keyboard/
cd msi-keyboard
rm -rf node_modules
npm i
sudo node examples/color.js
```
I highly doubt you need the signal11 hid module, your OS should ship with it along with libusb. I tested the project on mint 17 and currently on Debian 8.
BTW I would recommend not installing the project as global (-g) as it makes it harder to have multi copies for different projects.

bash64, 2016-06-30:
$ node -v
v4.2.6

bash64, 2016-06-30:
That worked. Now to figure out why.
I think its the global switch.

bash64, 2016-06-30:
I pull down 3 pieces of code.
Do I need them all still or just msi-keyboard?

>git clone https://github.com/wearefractal/msi-keyboard
>git clone https://github.com/hanshuebner/node-hid
>cd ~/node-hid
>git clone https://github.com/signal11/hidapi.git

My rainbow.js is the same as your color.js but I cannot run mine. It seems that the formatting requirements have changed for the js code:

Mine:
var keyboard = require('/usr/local/lib/node_modules/msi-keyboard');
keyboard.color('left',{color: 'red', intensity: 'med'});
keyboard.color('middle',{color: 'blue', intensity: 'med'});
keyboard.color('right',{color: 'green', intensity: 'med'});

Yours:
var keyboard = require('../')();

keyboard.color('left', {
	color: 'red',
	intensity: 'high'
});
keyboard.color('middle', {
	color: 'green',
	intensity: 'high'
});
keyboard.color('right', {
	color: 'blue',
	intensity: 'high'
});

bash64, 2016-06-30:
I got it all to work by using your examples only.
Thanks for the help.

Is there a list of msi laptops this works on?
Does it work on any external keyboards from msi?

stevelacy, 2016-06-30:
This is only designed for built in MSI Steel Series keyboards, it works on all since the first version released.

closed 2016-06-30 by stevelacy

bash64, 2016-06-30:
Do I need the line "sudo ln -s /usr/bin/nodejs /usr/bin/node"?
I included that to avoid errors as ubuntu renamed node to nodejs.

stevelacy, 2016-06-30:
if you installed node from apt-get then yes. if you installed it from source then no