stephenlacy/msi-keyboard #52: Feature Request: Change key color on keypress

dmk255, 2016-11-06:
I'd love to see the ability to change specific key's color on keypress. A couple examples of this would be for numlock, scroll lock, caps lock, etc. 

I would write the hook to execute a particular command on keypress myself, but at the moment there isn't a command to change a specific key's color (so far as I'm aware).

stevelacy, 2016-11-07:
Change each individual key's LED on that key's press?

dmk255, 2016-11-07:
Yup! So if I press my caps lock key I can change the LED color to blue. On
the second press it would revert to the default color for the region.

stevelacy, 2016-11-07:
Unfortunately this is not possible with stealseries keyboards, the keys are in regions not individually controllable.

closed 2016-12-01 by stevelacy