stephenlacy/msi-keyboard #54: Control additional regions for GT72 and (possibly?) newer models

JuanCAlpizar, 2017-01-04:
On the GT 72 and newer models, there's a new set of regions that can be colored that weren't available on previous models, resulting on a lot of lights off using the current implementation. I made a way to work with all of them and did some changes to code and this is a continuation from the discussion on #43 . The full commit can be checked out on my forked project here https://github.com/JuanCAlpizar/msi-keyboard/commit/0d7d07ae18788777c3d1a8f6654b1e404ea0876a.

Additional regions can be added to the current array with the proper mapping, like this:

 	"regions": {
		"left": 1,
		"middle": 2,
		"right": 3,
		"logo": 4,
		"front-right-panel": 5,
		"front-left-panel": 6,
		"trackpad": 7
	}

After that I made some additional changes, but those are merely suggestions. First I completely halted the execution if no proper regions are specified for breathe and wave modes like this:

 	if (typeof regions === 'undefined'){ // if no regions are specified at all, stop the process
		console.log('Error: no regions to work have been specified');
		return false;
	}

Then I changed how the default object is filled if a default color is the only thing specified on the region parameter.

 	if (typeof regions !== 'object'){
		var default_color = regions;
		regions = constants.regions;
		for (var region_name in constants.regions){
			regions[region_name] = default_color;
		}
	}

Finally I slightly restructured the cycle to apply colors and mode to regions, basically because there's no need of an array inside an array to determinate the color and region to color and there's no need to parse the index in every iteration. This shouldn't affect execution time to be noticeable, but sometimes I'm an optimization freak :p 

 	var region_layers = {
		'foreground_color_layer': 0,
		'background_color_layer': 0,
		'execution_period_layer': 0
	};
	var Def = null;
	for (var region_to_color in regions){
		if (typeof regions[region_to_color] !== 'undefined'){
			Def = getSettingsForRegion(regions[region_to_color], region_to_color);
			sendData (keyboard, region_layers.foreground_color_layer + 1, Def.primary.color, Def.primary.level, 0);
			sendData (keyboard, region_layers.background_color_layer + 2, Def.secondary.color, Def.secondary.level,
			sendData (keyboard, region_layers.execution_period_layer + 3, period, period, period);
		}else{
			sendData (keyboard, region_layers.foreground_color_layer + 1, {redScale: 0, greenScale: 0, blueScale: 0}, 0, 0);
			sendData (keyboard, region_layers.background_color_layer + 2, {redScale: 0, greenScale: 0, blueScale: 0}, 0, 0);
			sendData (keyboard, region_layers.execution_period_layer + 3, period, period, period);
		}
		
		region_layers.foreground_color_layer += 3;
		region_layers.background_color_layer += 3;
		region_layers.execution_period_layer += 3;
		
	}

Please let me know if there's a way we can merge this into your project, and thanks so much for making the hard work possible! 👍

stevelacy, 2017-01-04:
Thanks, quick question, how did you find out what hex headers to send to the device? https://github.com/JuanCAlpizar/msi-keyboard/commit/0d7d07ae18788777c3d1a8f6654b1e404ea0876a#diff-08b9a516c0639492957e3ab9b25f5ed9R211

JuanCAlpizar, 2017-01-04:
At first sadly I couldn't find the proper feature report blobs using node-hid, then I checked this C++ implementation, which helped me to better understand how the keyboard regions and layers worked https://gist.github.com/makkarpov/b559a82724158adeecf195e26ef3d951 after noticing he was using different hex headers and he was indeed able to control rbg values, I assumed there should be a different hex header with a closer value to your implementation to do so. After that it was trial and error until I found the proper hex header for set color and gradients

dimitriscsd, 2017-01-19:
This is great. I'm a GT72 user and I'd love to get this working

JuanCAlpizar, 2017-01-25:
@dimitriscsd the fork in my repository has the basics on the API working, would you mind to check if that does work for you too? I also have an MSI GT 72

dimitriscsd, 2017-01-25:
I'll give it a shot when i manage to find some spare time

abeagley, 2017-01-26:
@JuanCAlpizar Looks like your fork is working for my new MSI GT73VR-7RF on Fedora 25. I'm able to setup my track pad lighting successfully! Great job.

JuanCAlpizar, 2017-01-27:
@abeagley nice! are the regions matching though? Looks like the new GT 73/83 have once again a different layout

abeagley, 2017-01-27:
The regions seem to be working. I'm curious if they potentially added more to smooth out the gradient for newer models? There's also a light on the back of the laptop similar to Apple that I need to get figured out.

stevelacy, 2017-01-27:
The light on the back is powered by the screen brightness, it is just a semi-transparent plastic.

stevelacy, 2017-10-17:
Can you check if 9d173783a4dec43efce1a93695491193f6b448fc added the regions for you?

abeagley, 2017-10-21:
@stevelacy all but the 'touchpad' region are lighting up with your new commit on the GT73VR.

z1an0124, 2017-10-21:
Im using GT72 2QE laptop and i just switched over linux with a fresh install on linux mint (im a beginner) may i ask for a step by step guidance to install steel series for colours? (i apologize in advance in case this is not for a linux version)

z1an0124, 2017-10-21:
Actually i just found that i'm able to make work the led on the keyboard by just pressing the side button but not on the sides and on touch pad!

stellarpower, 2017-11-24:
Sorry, I messed up the touchpad constant, I've now fixed it. I made the change in my local sandbox and forgot to commit it. It's now working for me.

closed 2022-05-02 by stevelacy