stephenlacy/msi-keyboard #57: Fails Silently on GT72s Dominator Pro Dragon Edition

stellarpower, 2017-09-06:
Hi,

Not sure what I've done wrong here, I've cloned and used npm install and am running the basic set colour testscript as root, the program returns successfully but nothing happens on my keyboard. I've tried pressing the kill switch but it doesn't seem to do anything. I'm new to node.js but have installed the 'Node Inspector' debugger and can have a look at what's happening when I have a bit more time. Please could you tell me, do you know what the issue could be?

Thanks

stellarpower, 2017-09-06:
Apologies, nearly forgot. I'm on Mint Sonya (18.2) AMD64.

stevelacy, 2017-09-06:
There isn't really any debug method on the node side, it is probable that the keyboard is unsupported.

Could you get the output from this? https://github.com/stevelacy/node-hid-test
Is there a way to turn the backlight on with a keyboard key before running the program?
Can you try the GUI compiled binary? https://github.com/stevelacy/msi-keyboard-gui

stellarpower, 2017-09-06:
Hi,

Output of hid test is as follows:

[ { vendorId: 6000,
    productId: 65280,
    path: '0001:0002:00',
    release: 272,
    interface: 0 } ]

The keyboard's device ID from lsusb is the same as in the program, 1770:ff00 as you can see from above. The keyboard light button doesn't seem to do anything at all, is it a pure hardware button? I tried the GUI originally and similarly, nothing happened with no error. Should I try a live CD for a different distro to see if that's the issue?

stellarpower, 2017-09-06:
I see on the tested OSs, node is v6, node --version outputs 4.2.6 but it should be the latest package. Are these different version numbers or am I on an old build?

stevelacy, 2017-09-06:
node 6+ is required

stellarpower, 2017-09-06:
Updated to node 8, same as before.

stellarpower, 2017-09-06:
How did you work out the exact constants needed in the header? Could some experimentation with this be useful?

stevelacy, 2017-09-07:
Reverse engineered it along with looking through a few other earlier c and c++ attempts floating around github.

You should be able to enable the leds power switch with some keyboard button or combination. The ghost series require the leds to be powered on with a hardware switch first.

stevelacy, 2017-09-07:
I looked at an unboxing video for your laptop (probably the same year).
Hold `fn` and press the numpad minus `-` button.

stellarpower, 2017-09-07:
Excellent, that's it! Thank you so much. I had thought that those were the screen brightness and when it didn't change anything assumed that either they weren't recognised in Linux or they were for contrast. I must have turned the LEDs all the way off so that the keyboard LED button would no longer do anything.

Sorry for my incompetence and thank you very much!!

closed 2017-09-07 by stellarpower

stevelacy, 2017-09-07:
No worries, glad to help!
How do the projects work for you?