stephenlacy/msi-keyboard #61: How to determine new configurations / support for MSI GE63VR

redkite, 2017-12-15:
I tried to get this project (and other C++ siblings) running on my GE63VR SteelSeries laptop, so far without success.

In the normal configuration the usb vendor and device id are off, so that's already an indicator something has changed, the hex ID's are now `1038:1122`. But even if I change those I will get no feedback from the keyboard.

The backlighting itself works very well, it keeps changing colors in a flow manner all the time, I don't know what the mode is called. I never booted Windows since I need the laptop for work, but be very glad to have a constant backlight on the keyboard.

So, my key question is, to determine for myself how I can work out the correct code to send to the USB device, if it has changed. Where did you get your original information about this? Although I'd prefer a lightweight C++ implementation I would be happy to share any results (if I can produce any).

stevelacy, 2017-12-15:
Honestly from a 5+ year old c++ snippet floating somewhere on the interwebs.
The usb-hid was the easier part as the hid.devices() will return the correct hardware. Getting the correct constants for changing the serial bus was more complex, and was mostly based from the c++ snippet.
You should be able to swap out the vender id and product id directly in this project and get it to run assuming that they have a usb-hid interface. If not they must also update the dedicated windows SteelSeries engine, and that could give insights (with reverse engineering).

redkite, 2017-12-15:
Swapping out the vendor id was the easy part, but it doesn't really make a difference, some part of the setup has to be different. So I really have to find a Windows machine to get the software installed? Then I'll try that, thanks. Maybe I can come up with a solution.

siemovit, 2018-03-13:
redkite tell me if you finally got a solution, I am running on the same GE63VR SteelSeries.

richardnixon, 2018-05-07:
Has anyone been able to get this working on a GE63VR?

Pwpon500, 2018-05-08:
I've got the same problem. I was able to swap out the vendor ID and product ID, but when I run it, I just get this error:

``` none
node: ../../libusb/io.c:2118: handle_events: Assertion `ctx->pollfds_cnt >= internal_nfds' failed.
Aborted
```

eduardoleal, 2018-07-24:
@Pwpon500 I'm having the exact same output. Were you able to solve that?

Pwpon500, 2018-08-06:
@eduardoleal I was not. I'm not familiar with how USB IDs work, so I'm not quite sure what the problem is.

closed 2022-05-02 by stevelacy