stephenlacy/msi-keyboard #62: newest code not working

bash64, 2018-03-27:
on my msi gt70-2pe laptop:

in the examples folder the color.js script no longer works.
I used to get a rainbow colored solid keyboard.
Now I get lightblue single color.
I recovered color.js from a backup. The code is different. see below.
The older color.js does not work anymore.

I normally login to my desktop and run "sudo nodejs color.js". It is important to me to see in the dark.

Breathe and wave still work (newest copies).

I also use a custom script called off.js that sets all 3 colors to black.
It now does nothing.

If I figure it out I will close the issue.

-------------------------
the old color.js
-----------------------
var keyboard = require('../')();

keyboard.color('left', {
	color: 'red',
	intensity: 'high'
});
keyboard.color('middle', {
	color: 'green',
	intensity: 'high'
});
keyboard.color('right', {
	color: 'blue',
	intensity: 'high'
});

the new color.js
-------------------------
var keyboard = require('../')();

keyboard.color('left', {
	color: 'rgba(200, 0, 0)',
	intensity: 'high'
});
keyboard.color('middle', {
	color: 'green',
	intensity: 'high'
});
keyboard.color('right', {
	color: '#4654BD',
	intensity: 'high'
});

bash64, 2018-03-27:
p.s. do i need to install node-hid anymore? seems to be a part of msi-keyboard.

here are my install instructions that i currently use:
>sudo apt-get install git libusb-dev libusb-1.0-0-dev nodejs npm build-essential --install-recommends
>sudo ln -s /usr/bin/nodejs /usr/bin/node
>git clone https://github.com/stevelacy/msi-keyboard
>git clone https://github.com/node-hid/node-hid
>cd ~/msi-keyboard;  npm i
>cd ~/node-hid;  npm i

stevelacy, 2018-03-27:
You do not need to do that extra node hid clone, just clone the main `msi-keyboard` repo and run `npm i`. It was never needed as a install step, not sure where you read that.

You also don't need to go into backups to get an earlier version of the project, you can check the releases to find them: https://github.com/stevelacy/msi-keyboard/releases

Try the `examples/flow.js` You will need to install color and sleep in the examples folder.

bash64, 2018-03-27:
I have removed the extra node-hid from my instructions.
Here is the picture of the light blue single color produced by running "sudo nodejs color.js":

Any reason for this?

[image]

bash64, 2018-03-29:
I have recovered my old msi-keyboard folder (0.4) and all is working. I will stay with that version until the newer version starts working for me. I do not know what changed, but it doesn't work on my laptop.

stellarpower, 2018-04-09:
You need to install the node package color I think, sorry, I added that to the code but didn't change the dependencies, I don't use node normally so didn;t do anything to the package. The colorpackage enables rgb colours not just predefined ones.

bash64, 2018-04-09:
Is it a package called colorpackage or is it github code?

bash64, 2018-04-09:
I did the following:
>cd msi-keyboard
>npm i color

issue is fixed.
the new code works.

closed 2018-04-09 by stevelacy