stephenlacy/msi-keyboard #63: Error on New Version of NODE_MODULE_VERSION Needed

raedeus, 2018-05-07:
I am now generating this error when I try to run "node example/color.js":

Error: The module '/node_modules/node-hid/build/Release/HID.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 57. This version of Node.js requires
NODE_MODULE_VERSION 64. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:707:18)
    at Module.load (internal/modules/cjs/loader.js:589:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:528:12)
    at Function.Module._load (internal/modules/cjs/loader.js:520:3)
    at Module.require (internal/modules/cjs/loader.js:626:17)
    at require (internal/modules/cjs/helpers.js:20:18)
    at bindings (/home/******/node_modules/bindings/bindings.js:81:44)
    at Object.<anonymous> (/home/******/node_modules/node-hid/nodehid.js:5:34)
    at Module._compile (internal/modules/cjs/loader.js:678:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:689:10)

When I try to rebuild in the msi-keyboard folder I get the following:

******@ghost:[msi-keyboard] $ npm install

> node-hid@0.5.7 install /home/******/build/msi-keyboard/node_modules/node-hid
> prebuild-install || node-gyp rebuild

prebuild-install info begin Prebuild-install version 2.4.1
prebuild-install info looking for local prebuild @ prebuilds/node-hid-v0.5.7-node-v64-linux-x64.tar.gz
prebuild-install info looking for cached prebuild @ /home/******/.npm/_prebuilds/https-github.com-node-hid-node-hid-releases-download-v0.5.7-node-hid-v0.5.7-node-v64-linux-x64.tar.gz
prebuild-install http request GET https://github.com/node-hid/node-hid/releases/download/v0.5.7/node-hid-v0.5.7-node-v64-linux-x64.tar.gz
prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v0.5.7/node-hid-v0.5.7-node-v64-linux-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=10.0.0 runtime=node arch=x64 platform=linux)
make: Entering directory '/home/******/build/msi-keyboard/node_modules/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
../hidapi/libusb/hid.c: In function ‘hid_read_timeout’:
../hidapi/libusb/hid.c:1078:6: warning: variable ‘bytes_read’ might be clobbered by ‘longjmp’ or ‘vfork’ [-Wclobbered]
  int bytes_read = -1;
      ^~~~~~~~~~
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
In file included from ../src/HID.cc:31:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:840:60: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::Function>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, func, argc, argv);
                                                            ^
In file included from ../src/HID.cc:30:0:
/home/******/.node-gyp/10.0.0/include/node/node.h:172:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/******/.node-gyp/10.0.0/include/node/node.h:88:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/HID.cc:31:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:855:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, v8::Local<v8::String>, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, symbol, argc, argv);
                                                              ^
In file included from ../src/HID.cc:30:0:
/home/******/.node-gyp/10.0.0/include/node/node.h:165:50: note: declared here
                 NODE_EXTERN v8::Local<v8::Value> MakeCallback(
                                                  ^
/home/******/.node-gyp/10.0.0/include/node/node.h:88:42: note: in definition of macro ‘NODE_DEPRECATED’
     __attribute__((deprecated(message))) declarator
                                          ^~~~~~~~~~
In file included from ../src/HID.cc:31:0:
../../nan/nan.h: In function ‘v8::Local<v8::Value> Nan::MakeCallback(v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’:
../../nan/nan.h:870:62: warning: ‘v8::Local<v8::Value> node::MakeCallback(v8::Isolate*, v8::Local<v8::Object>, const char*, int, v8::Local<v8::Value>*)’ is deprecated: Use MakeCallback(..., async_context) [-Wdeprecated-declarations]
         v8::Isolate::GetCurrent(), target, method, argc, argv);
                                                              ^
In file included from ../src/HID.cc:30:0:
[... 59 lines cut ...]
make: Leaving directory '/home/******/build/msi-keyboard/node_modules/node-hid/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/node-gyp/lib/build.js:258:23)
gyp ERR! stack     at ChildProcess.emit (events.js:182:13)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:225:12)
gyp ERR! System Linux 4.16.6-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/******/build/msi-keyboard/node_modules/node-hid
gyp ERR! node -v v10.0.0
gyp ERR! node-gyp -v v3.6.2
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@0.5.7 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@0.5.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Any input would be helpful.

Thanks,

raedeus

raedeus, 2018-05-07:
Hey there! The second grouping of output is the result of that. Yes. Scratching my head.

stevelacy, 2018-05-07:
Can you try with a node 8/9 version? Not sure if node-hid has issues with node 10

raedeus, 2018-05-07:
Sure thing. I will report back with my findings.

Thanks for your assistance with this!

raedeus

raedeus, 2018-05-07:
Yep. Pulled down node 8 and ran with that just fine. I guess it's a problem with 10. I will just run with that binary until it get's fixed.

So, maybe this is a bug report then?

- raedeus

stevelacy, 2018-05-07:
Try to clone just node-hid and run their example on node 10.

raedeus, 2018-05-07:
Will do when I return end of day and report back.

Thanks,

- raedeus

stevelacy, 2018-05-08:
Yeah sounds like a node 10 issue with node-hid...

tarach, 2018-06-28:
Please share exact version of node that worked for you.

I've tried 8 different versions already and I'm getting **This version of Node.js requires
NODE_MODULE_VERSION 64 or 57** all the time. ~thanks

raedeus, 2018-06-28:
I will forward the version as soon as I get back on the system. Currently
away. ETA end of day today PDT.

Thanks.

raedeus, 2018-06-28:
Hey tarach,

node-v8.11.1

So I didn't screw anything up I compiled it, left it in the bin directory and ran it as such [absolute paths removed]:

[...]/node-v8.11.1-linux-x64/bin/node [...]/msi-keyboard/color.js

tarach, 2018-06-30:
Shortly after posting I found out that my MSI GL62 7RD does not support keyboard backlight. As I remember I decided its pointless to spend another 100 $ for a feature that most likely is not supported on linux... Well I was clearly wrong.

Either way. Just to let you know, after changing version to 8.11.1 ( npm 5.6.0 ) I'm still getting the same messages.

*msi-keyboard-gui*:
> The module '/home/sebastian/projects/msi-keyboard-gui/dist/app/node_modules/node-hid/build/Release/HID.node'
> was compiled against a different Node.js version using
> NODE_MODULE_VERSION 54. This version of Node.js requires
> NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
> the module (for instance, using `npm rebuild` or `npm install`).

*msi-keyboard*:
> $ which node
> /root/.nvm/versions/node/v8.11.1/bin/node
> $ node examples/color.js 
> /home/sebastian/projects/msi-keyboard/node_modules/node-hid/nodehid.js:24
>     this._raw = new (Function.prototype.bind.apply(binding.HID,
>                 ^
> 
> Error: cannot open device with vendor id 0x1770 and product id 0xff00
>     at new HID (/home/sebastian/projects/msi-keyboard/node_modules/node-hid/nodehid.js:24:17)
>     at module.exports (/home/sebastian/projects/msi-keyboard/lib/findKeyboard.js:8:14)
>     at Object.<anonymous> (/home/sebastian/projects/msi-keyboard/examples/color.js:1:92)
>     at Module._compile (module.js:652:30)
>     at Object.Module._extensions..js (module.js:663:10)
>     at Module.load (module.js:565:32)
>     at tryModuleLoad (module.js:505:12)
>     at Function.Module._load (module.js:497:3)
>     at Function.Module.runMain (module.js:693:10)
>     at startup (bootstrap_node.js:188:16)

stevelacy, 2018-07-02:
The second command needs to be run as root, `sudo node examples/color.js`
As for the first error, use a prebuilt binary from the latest releases.

stevelacy, 2018-07-02:
@tarach if your laptop does not have the stealseries keyboard then this module will always fail with 'Cannot open device'

tarach, 2018-07-02:
I did call it as root. I had to switch to root account instead of using sudo because I'm using nvm and it wouldn't work with sudo.

My laptop does have stealseries keyboard but without backlight. 
https://msi.com/Laptop/GL62-7RD.html

closed 2022-05-02 by stevelacy