stephenlacy/msi-keyboard #65: Does it support MSI WE73?

stilor, 2019-01-10:
It has the same USB device ID for the keyboard (1770:ff00) but setting any color does not seem to have any effect. Would love to be able to turn the backlight on/off...

stevelacy, 2019-01-10:
That is odd, if it has the same product/manufacturer HID ids then _it should_ work.
There are two keyboards, the older 2010ish - 2016/17 and the newer single key activator.

Is this it? https://us.msi.com/Workstation/WE73.html
That laptop does not seem to have colored backlit keys

stilor, 2019-01-10:
Yes, it is that laptop. It doesn't have colored backlight but it has backlight that can be switched on/off (at least in Windows, the profiles (in "MSI Dragon Center") can configure whether backlight is enabled.

stevelacy, 2019-01-10:
The only command that might work is the full on/off toggle, setting each region to 'black'

stilor, 2019-01-10:
That's what I tried. Unfortunately, it didn't have any effect. I guess command codes are different for white-only backlight.

closed 2022-05-02 by stevelacy