stephenlacy/msi-keyboard #71: cannot open usb device error in MSI GP66 model

likith3611, 2022-07-10:
Hii I installed msi-keyboard and when I ran the command as mentioned it gave me an error something like this
[image]

I installed lib packages mentioned and also tried with sudo but im facing the same issue can you please help me with this (PS It was flawlessly in windows)

likith3611, 2022-07-10:
[image]

stevelacy, 2022-07-10:
Sounds like you are missing some usb library for your linux OS if this was working in windows

likith3611, 2022-07-13:
Hmm I tried installing libusb and libusb-dev is there anything else?

likith3611, 2022-07-13:
I tried to install usb drivers and also did sudo ubuntu-drivers autoinstall so that I would get updated list of drivers but still stuck with the same issue any idea what can be done?

likith3611, 2025-01-02:
Closing this issue and figured out the drivers officially and unoffically arent supported for GP66 thanks for all the help!

closed 2025-01-02 by likith3611